vlmaps/vlmaps #2: Building the VLMap for real world environment

AbdelrahmanYounes, 2023-03-02:
Hey Chenguang,

Congrats on your great work and on its acceptance in ICRA!

I'm trying to build the Vlmap for our lab environment and I followed the steps you mentioned in the paper about collecting the data with RTAB-MAP, but when I do the same and have the 3 folders of (rgb, depth, pose) with the converted the depth.png images to depth.npy like the example and then trying to build the map It doesn't get build correctly like the image attached. So, could you please let me know if you had to tweak something for the real-world experiment? or what I might have done wrongly? Thanks in advance! 

[image]

Tom-Huang, 2023-03-02:
Hi Abdelrahman,
Thank you for your interest in our work. From your image, I suspect that the pose of the camera is inverted. Could you try inverting the transformation before building the map? If this is not the cause, I would appreciate it if you could provide your script and we can see what the problem is.

Tom-Huang, 2023-03-09:
You need to make some modifications (mainly the transformations) to make the jupyter code work on the real robot. The code in the jupyter notebook is for the simulator. The code assumes that the `pose` folder stores the robot base poses relative to the habitat simulator world frame (x pointing right, y pointing up, z pointing backward). It also assumes that the camera is pointing forward (no pan no tilt) and 1.5 meters above the base link, meaning that the camera frame relative to the robot base frame is:
```
tf_ro2cam = [
[1, 0, 0, 0],
[0, -1, 0, 0],
[0, 0, -1, 1.5],
[0, 0, 0, 1]
]
```

But RTAB-Map returns poses of the camera (x pointing right, y pointing down, z pointing forward) but not the base of the robot. You should convert the RTAB-Map camera poses into the robot base frame. If you use ROS, you can read the relative transform between the `base_link` and `your_camera_frame_id` with tf: `rosrun tf tf_echo base_link your_camera_frame_id` ([http://wiki.ros.org/tf#tf_echo](http://wiki.ros.org/tf#tf_echo)). With this relative pose, you can convert the camera poses in the pose folder into the poses of the base_link. 

Another thing you need to pay attention to is that the VLMap is built relative to the first frame of the camera frame in simulator. But on the real world robot, since we can not guarantee that the camera is pointing forward, the map should be relative to the first frame of the base_link frame.

DiTo97, 2023-06-23:
Hi @AbdelrahmanYounes,

Did you manage to reconstruct a real-world VL map following @Tom-Huang's guidelines?

For the past few days I have been trying to do the same following a similar setup to Habitat's in my laboratory (a Intel RealSense D435i RGB-D camera mounted on a vertical support along the known robot body frame) capturing the RGB-D + pose stream both at 10 and 30 FPS under different lightning conditions (natural at midday and artificial at midnight).

I made sure the poses are stored respecting the coordinate system that @Tom-Huang detailed above with the only difference being the camera height at 0.84 meters over the robot's CoM corresponding to a height above the ground of 1.5 meters. Unfortunately, the reconstructed bird's eye view VL map is quite underwhelming.

I will attach below the RGB-D stream as well as the individual reconstructions provided in the notebook (obstacles, colored map and VL map). All relevant hyper-parameters (cell size, grid resolution, sample rate) were left unchanged.

**Edit:** The files were too big, so I shared them on [WeTransfer](https://we.tl/t-x5HMqbQbgg).

The link above will expire in one week, but I can share them again if necessary.

Are there any glaring mistakes that you can spot?

Jessy-Huang, 2023-07-21:
> Hey Chenguang,
>
> Congrats on your great work and on its acceptance in ICRA!
>
> I'm trying to build the Vlmap for our lab environment and I followed the steps you mentioned in the paper about collecting [...]

have you solved the problem yet？ I met the same problem with you by using matterport3d dataset downloaded from the matterport3d website， need I do some pre-process to improve the dataset？

Tom-Huang, 2023-09-14:
See #15

closed 2023-09-14 by Tom-Huang