vlmaps/vlmaps #3: Experiments on ai2thor

SeungyounShin, 2023-03-26:
I am currently working on a project to generate a 2D map using the AI2-THOR simulator. I have implemented the 2d_vlmap.py script in the [vlmaps_ai2thor repository](https://github.com/SeungyounShin/vlmaps_ai2thor/tree/main).

While the generated map makes some sense, it seems to be missing some details or suffering from inaccuracies. I'm not sure if this is due to an issue in my implementation or if the animated scene in AI2-THOR inherently results in poor performance.

Here's a sample output of the generated map:

[image]

I would appreciate any guidance on identifying and resolving the issue. Is there something I might have missed in my implementation, or are there known limitations with AI2-THOR that could be affecting the map generation?

Please let me know if you have any suggestions or insights that could help improve the accuracy of the generated map. I'm open to exploring alternative approaches or making adjustments to the current implementation to address this problem.

Thank you for your assistance.

SeungyounShin, 2023-03-26:
I have been working on a project to generate a 2D map using the AI2-THOR simulator. I previously encountered some inaccuracies in the generated map when using a resolution of 128. However, after increasing the RGB-D resolution to 512, the generated map has significantly improved in quality and closely matches the ground truth.

Here's a sample output of the generated map with the higher resolution:

[image]

closed 2023-03-26 by SeungyounShin

Tom-Huang, 2023-03-26:
Hi @SeungyounShin, thank you for your interest in our project. You are right, the resolution of the map (cell_size) can affect the language indexing accuracy especially when it comes to small objects. In the end, the features at every top-down grid cell contain the features of projected points lying in the same cell_size * cell_size area. If the resolution is higher (cell_size is smaller), the chance that the cell_size * cell_size region contains features from different objects is smaller. Therefore, the map is also more accurate.

SeungyounShin, 2023-03-27:
I am also curious as to why you have opted to project in two dimensions. It seems that using a voxel map for object navigation could potentially provide greater accuracy.

Tom-Huang, 2023-03-27:
yeah, it's nice that you also discover that. A 3D version of VLMaps would be better in accuracy, which was noticed by us later. It was used in our recent work AVLMaps. Feel free to check our new project [avlmaps.github.io](avlmaps.github.io). A 3D VLMaps implementation is also provided in that project's [CoLab page](https://colab.research.google.com/drive/1gdtLvg_Fbl16N3ITp5FsU9ZAG6HmspVb?usp=sharing)

DiTo97, 2023-05-03:
Hi @Tom-Huang,

Is there a timeline for the release of the code of audio visual language maps ([AVLMaps](https://avlmaps.github.io/))?

As you and @SeungyounShin were mentioning, the 3D voxel grid seems quite the step forward

mees, 2023-05-03:
Hey guys, we are trying our best to refactor the code to offer the best possible experience. We have some other responsibilities to take care of too right now, but we estimate at the latest to have everything up by the end of June hopefully.