vlmaps/vlmaps #4: Off the shelf navigation stack

sairitwik27, 2023-04-16:
I was going through the vlmaps paper, and was trying to find the off the shelf navigation stack that converts code as policies part of the code to habitat smulator. I couldn't find it in the github repository, are there any plans to release the code for it to be able to reproduce the results?

Tom-Huang, 2023-04-17:
I appreciate your interest in our paper. We are refactoring the code now. We will release the code soon.

juexZZ, 2023-10-24:
Hi @Tom-Huang @mees,
Just want to follow up on this. Are you still planning to share the codes for navigation? 
This is a very interesting work and many colleagues in my research community would love to play with it and cite your paper if we can have access to the navigation stack. :)

Tom-Huang, 2023-11-02:
we just release the navigation stack the the rest of the dataset. Feel free to try it out!

closed 2023-11-02 by Tom-Huang