vlmaps/vlmaps #6: 如何构建自己的vlmaps

Jessy-Huang, 2023-06-21:
您好，我在尝试根据您的论文构建自己的vlmaps，您论文中提及您也HSR mobile robot上实现了vlmap导航，我感觉您用的也是ros系统。我使用ros系统尝试建立vlmap，目前有几个疑问：
1. 您的depth下面的.npy文件转成array是一维的，正常的深度图片是二维的
2. pose文件中的txt文件，这是机器人的什么信息
3. 语义分割和对应的标签，是需要单独用自己的语义分割模型进行图片的处理吗

vpnasdfghjkl, 2023-07-05:
我刚读过这篇论文和代码可以回答一下：
ans1：depth文件里如果是一维的只需要reshape成图片的长和宽；一般使用类似astra这类深度相机处理深度数据的时候是可以直接处理成一维或者二维都可以的；
ans2：pose中的txt文件，前三个是机器人空间坐标；后四个是四元数表示相机的旋转情况；
ans3：这个论文里提到了LSeg，这个可以对rgb图片进行语义分割；

Jessy-Huang, 2023-07-18:
您好，我已经根据论文的步骤做好了数据集，但是得到的结果效果太差，不知道您有没有做过相关研究，不知道自己问题出现在哪里，如果有的话可否分享一下经验

从 Windows 版邮件发送

发件人: ForClever
发送时间: 2023年7月5日 20:39
收件人: vlmaps/vlmaps
抄送: Jessy-Huang; Author
主题: Re: [vlmaps/vlmaps] 如何构建自己的vlmaps (Issue #6)

我刚读过这篇论文和代码可以回答一下：
ans1：depth文件里如果是一维的只需要reshape成图片的长和宽；一般使用类似astra这类深度相机处理深度数据的时候是可以直接处理成一维或者二维都可以的；
ans2：pose中的txt文件，前三个是机器人空间坐标；后四个是四元数表示相机的旋转情况；
ans3：这个论文里提到了LSeg，这个可以对rgb图片进行语义分割；

vpnasdfghjkl, 2023-07-24:
请问你的数据集是怎么得来的，我曾使用深度相机手动调整过深度相机的的姿态，并且手动设置了pose.txt。共使用了4帧图片。1.首先场景不要太复杂（例如桌面上很多东西）。2.相机高度要高点，采样范围大概10*10m。3.相机的位姿变换采用的是右手坐标系，需要使用rad2quat转换为四元数。

Jessy-Huang, 2023-07-24:
我是用的ros机器人，搭载了深度相机，获取了rgb和depth，然后获取了pose信息，我的相机高度大致在0.5m，方便的话我们可以加微信交流，15579663430

从 Windows 版邮件发送

发件人: Deposit
发送时间: 2023年7月24日 10:01
收件人: vlmaps/vlmaps
抄送: Jessy-Huang; Author
主题: Re: [vlmaps/vlmaps] 如何构建自己的vlmaps (Issue #6)

请问你的数据集是怎么得来的，我曾使用深度相机手动调整过深度相机的的姿态，并且手动设置了pose.txt。共使用了4帧图片。1.首先场景不要太复杂（例如桌面上很多东西）。2.相机高度要高点，采样范围大概10*10m。3.相机的位姿变换采用的是右手坐标系，需要使用rad2quat转换为四元数。

Tom-Huang, 2023-09-14:
See #15

closed 2023-09-14 by Tom-Huang