vlmaps/vlmaps #8: 真实世界构建vlmaps效果太差

Jessy-Huang, 2023-07-17:
我在真实世界中，使用ros小车采集了实验室的环境，最终构建得到的vlmaps效果太差，猜测原因出现在获得obstacles图时得到的障碍物的坐标太少，在图片中只有零星的几个点，我修改了相机的高度，图片的大小采用了648*480，其余的都没有动，可以帮我解答一下问题出现在哪里嘛，数据集的进行了几次优化，但是效果就是不理想，感激不尽
[image]

Tom-Huang, 2023-07-19:
What sensors are you using? If you are using depth camera, you can first try to build a top-down obstacle map with depth images to verify that you have the correct transformation. Later you can try with back projecting the pixel features with the depth and accumulate the features in the top-down map.

Jessy-Huang, 2023-07-20:
I am using an ROS robot car. I collected depth, rgb, and pose information on the ROS car, and then based on your dataset, converted the depth and images trained with LSEG into npy and placed them in the model for training. Following the steps in your demo.ipynb file, I ran the code under Visualize the dataset and found that the depth video effect was poor, with many messy patterns, resulting in an observer map. However, the effect was poor, There are only scattered points, and no shape of any object can be seen. I compared your dataset with the original dataset of matterport3d, and found that there is a big difference between the two. I think you have processed rgb, and then the pose has also been transformed. I want to know how you processed the dataset. Can I use my own dataset to achieve better results with the same dataset? Here is my dataset information:

Tom-Huang, 2023-07-25:
>converted the depth and images trained with LSEG into npy and placed them in the model for training

We didn't train the LSeg but use the pre-trained LSeg model. You can directly apply [https://github.com/vlmaps/vlmaps/blob/bc79b26a577e5a9408f86e45e5c16530ca80f867/examples/clip_mapping_lseg_from_scratch_batch.py#L204](https://github.com/vlmaps/vlmaps/blob/bc79b26a577e5a9408f86e45e5c16530ca80f867/examples/clip_mapping_lseg_from_scratch_batch.py#L204) to an image to get the pixel-wise embedding.

>found that the depth video effect was poor, with many messy patterns, resulting in an observer map. However, the effect was poor, There are only scattered points, and no shape of any object can be seen.

You can use the depth to build a top-down obstacle map first to see if you get the poses correctly.

>I compared your dataset with the original dataset of matterport3d, and found that there is a big difference between the two. I think you have processed rgb, and then the pose has also been transformed

Yes, the poses in my example dataset are from Habitat simulator which are poses of the robot. But the poses in Matterport3D are poses of the camera. So you need to apply some transforms to make the poses usable. You can also see from the comment [here](https://github.com/vlmaps/vlmaps/blob/bc79b26a577e5a9408f86e45e5c16530ca80f867/examples/clip_mapping_lseg_from_scratch_batch.py#L127) that the poses in Habitat sim has `z backward, y upward, x to the right` but the Matterport3D dataset poses have `z forward, x rightward, y downward`. A quick solution you can try is applying following to your poses:
```python
robot_to_cam_transform = np.eye(4)
robot_to_cam_transform[2, 3] = height_of_cam_to_ground
robot_to_cam_transform[1, 1] = -1
robot_to_cam_transform[2, 2] = -1

robot_pose = your_cam_pos @ np.linalg.inv(robot_to_cam_transform)
```
here, `robot_pose` is the poses you use (or Matterport3D poses), `height_of_cam_to_ground` is the height of your camera on the robot car to the ground. After applying such transform to your poses, try with my code again to see if it works.

Tom-Huang, 2023-09-14:
See #15