vlmaps/vlmaps #9: 使用matterport3d原始数据集效果依旧很差

Jessy-Huang, 2023-07-21:
I used the dataset used in the paper, the one starting with 5L, and compared it with the dataset provided by matterport3d. It seems that you have processed the data, and there is a significant difference between your dataset and the dataset provided by matterport3d. Based on your first response, I converted the position coordinates and used lseg to perform semantic segmentation on the rgb image. The other parameters were not changed, but the final result was very poor. Can you provide me with some ideas, What exactly is the problem? I have been stuck in this place for almost a month and the results of using my own dataset and matterport3D dataset are very poor. Thank you very much!!!
[image]

Tom-Huang, 2023-07-25:
I guess the poses are not correct. The poses in Matterport3D dataset are poses of the camera, which are not the same as the poses from the Habitat simulator (poses of the robot). If you want to make it work on Matterport3D directly, you need to modify the poses processing below.
[https://github.com/vlmaps/vlmaps/blob/bc79b26a577e5a9408f86e45e5c16530ca80f867/examples/clip_mapping_lseg_from_scratch_batch.py#L126-L132](https://github.com/vlmaps/vlmaps/blob/bc79b26a577e5a9408f86e45e5c16530ca80f867/examples/clip_mapping_lseg_from_scratch_batch.py#L126-L132)
So you are right, I didn't directly use Matterport3D but collected RGBD streams and poses from the Habitat simulator.

Tom-Huang, 2023-09-14:
See #15