vlmaps/vlmaps #15: The result of the coordinate transformation is still incorrect

Jessy-Huang, 2023-09-07:
I have made the correct coordinate transformation according to the instructions in the paper and your prompts in the issue. You can refer to the following link for the specific process:
https://developer.orbbec.com.cn/v/forum_detail/14557
[image]

and the dataset as follow：
https://iklxo6z9yv.feishu.cn/drive/folder/Yel4fN43Jl2pMGdfSSNc5vdxnLb?from=from_copylink

Tom-Huang, 2023-09-07:
could you provide the full code? And please continue the discussion in the same issue and don't open too many issues. Thanks a lot.

Jessy-Huang, 2023-09-07:
here is the full code in jupyter notebook。 Thanks a lot for answering my question
[demo_car.zip](https://github.com/vlmaps/vlmaps/files/12550844/demo_car.zip)

Jessy-Huang, 2023-09-08:
I'm not sure if you have received my reply message from GitHub. If you have time, could you please help me check where my code problem is? This is my complete code

Jessy-Huang, 2023-09-09:
I think I made a reasonable coordinate transformation and got a result that seemed a bit reasonable, but the map's effect was still too poor, possibly due to those reasons. Below are my documents related to coordinate transformation：
`# @markdown meters per cell size
cs = 0.05 # @param {type: "number"}
# @markdown map resolution (gs x gs)
gs = 500 # @param {type: "integer"}
# @markdown camera height (used for filtering out points on the floor)
# 摄像头相对于小车底盘升高43.5cm
camera_height = rate*(7500+700+4350) # @param {type: "number"}
# camera_height = rate*(4350) # @param {type: "number"}
# @markdown depth pixels subsample rate
depth_sample_rate = 5 # @param {type: "integer"}
# @markdown data where rgb, depth, pose are loaded and map are saved
data_dir = data_dir # @param {type: "string"}
print(data_dir)`

`# read pose
pos, rot = load_pose(pose_path)  
# z backward, y upward, x to the right  ros：y backward, z upward, x to the right
#         rot_ro_cam = np.eye(3)
#         rot_ro_cam[1, 1] = -1
#         rot_ro_cam[2, 2] = -1
#         rot_ro_cam = np.array([[0, 0, 1],
#                                [-1, 0, 0],
#                                [0, -1, 0]])
rot_ro_cam = np.array([[0, 0, 1],
                       [1, 0, 0],
                       [0, -1, 0]])
rot = rot @ rot_ro_cam
pos[2] += camera_height

pose = np.eye(4)
pose[:3, :3] = rot
pose[:3, 3] = pos.reshape(-1)

tf_list.append(pose)
if len(tf_list) == 1:
    init_tf_inv = np.linalg.inv(tf_list[0]) 

tf = init_tf_inv @ pose`

[image]

Jessy-Huang, 2023-09-09:
I think I made a reasonable coordinate transformation and got a result that seemed a bit reasonable, but the map's effect was still too poor, possibly due to those reasons. Below are my documents related to coordinate transformation

```
# @markdown meters per cell size
cs = 0.05 # @param {type: "number"}
# @markdown map resolution (gs x gs)
gs = 500 # @param {type: "integer"}
# @markdown camera height (used for filtering out points on the floor)
# 摄像头相对于小车底盘升高43.5cm
camera_height = rate*(7500+700+4350) # @param {type: "number"}
# camera_height = rate*(4350) # @param {type: "number"}
# @markdown depth pixels subsample rate
depth_sample_rate = 5 # @param {type: "integer"}
# @markdown data where rgb, depth, pose are loaded and map are saved
data_dir = data_dir # @param {type: "string"}
print(data_dir)
```

```
# read pose
pos, rot = load_pose(pose_path)  
# z backward, y upward, x to the right  ros：y backward, z upward, x to the right
#         rot_ro_cam = np.eye(3)
#         rot_ro_cam[1, 1] = -1
#         rot_ro_cam[2, 2] = -1
#         rot_ro_cam = np.array([[0, 0, 1],
#                                [-1, 0, 0],
#                                [0, -1, 0]])
rot_ro_cam = np.array([[0, 0, 1],
                       [1, 0, 0],
                       [0, -1, 0]])
rot = rot @ rot_ro_cam
pos[2] += camera_height

pose = np.eye(4)
pose[:3, :3] = rot
pose[:3, 3] = pos.reshape(-1)

tf_list.append(pose)
if len(tf_list) == 1:
    init_tf_inv = np.linalg.inv(tf_list[0]) 

tf = init_tf_inv @ pose
```

[image]

Tom-Huang, 2023-09-13:
By default, the camera coordinate is z forward, x right, y down but your x is pointing left which is incorrect. In addition, you should not use [depth2pc](https://github.com/vlmaps/vlmaps/blob/master/utils/clip_mapping_utils.py#L199) because this back projection function relies on the simulated intrinsics matrix [get_sim_cam_mat_with_fov](https://github.com/vlmaps/vlmaps/blob/master/utils/clip_mapping_utils.py#L419). You should use [depth2pc_real_world](https://github.com/vlmaps/vlmaps/blob/master/utils/clip_mapping_utils.py#L171) instead and pass the intrinsics matrix of your depth camera as `cam_mat` argument.

Jessy-Huang, 2023-09-14:
> ```
>     for data_sample in data_iter:
>         rgb_path, depth_path, semantic_path, pose_path = data_sample
>         
>         bgr = cv2.imread(rgb_path)
>         rgb = cv2.cvtColor(bgr, cv2.COLOR_BGR2RGB)
> 
>         pos, rot = load_pose(pose_path)  
>         rot_ro_cam = [[0, 0, 1],
>                       [-1, 0, 0],
>                       [0, -1, 0]]
>         rot = rot @ rot_ro_cam
>         pos[2] += camera_height
> 
>         # this is camera pose
>         pose = np.eye(4)
>         pose[:3, :3] = rot
>         pose[:3, 3] = pos.reshape(-1)
> 
>         tf_list.append(pose)
>         if len(tf_list) == 1:
>             init_tf_inv = np.linalg.inv(tf_list[0]) 
> 
>         # this is the base pose with the first base pose as the world frame (base's world frame)
>         tf = init_tf_inv @ pose
> 
>         # read depth
>         depth = rate*load_depth(depth_path)
> 
>         # read semantic
>         semantic = load_semantic(semantic_path)
>         semantic = cvt_obj_id_2_cls_id(semantic, obj2cls)
> 
>         pix_feats = get_lseg_feat(model, rgb, labels, transform, crop_size, base_size, norm_mean, norm_std)
>         
>         # transform all points to the global frame
>         # pc, mask = depth2pc(depth, 57.0)  # 奥比中光摄像头H-fov=57.0
>         
>         # TODO: put import to top of file
>         from utils.clip_mapping_utils import depth2pc_real_world
>         # TODO: define your intrinsic_matrix: 
>         # intrinsic_matrix = # (3, 3) matrix
>         pc, mask = depth2pc_real_world(depth, intrinsic_matrix)  # 奥比中光摄像头H-fov=57.0
>         shuffle_mask = np.arange(pc.shape[1]) 
>         np.random.shuffle(shuffle_mask)
>         shuffle_mask = shuffle_mask[::depth_sample_rate]
>         mask = mask[shuffle_mask]
>         pc = pc[:, shuffle_mask]
>         pc = pc[:, mask]
>         # this is the camera frame relative to the base's world frame
>         pc_global = transform_pc(pc, tf)
> 
>         # rgb_cam_mat = get_sim_cam_mat(rgb.shape[0], rgb.shape[1])
>         # TODO: define your intrinsic matrix
>         # rgb_cam_mat =  # (3, 3) matrix
>         feat_cam_mat = get_sim_cam_mat(pix_feats.shape[2], pix_feats.shape[3])
>         for i, (p, p_local) in enumerate(zip(pc_global.T, pc.T)):
>                  ...
> ```
> 
> try to replace the corresponding part of your code with this snippet and finish the `TODOs`. I checked your code but couldn't download your data as a whole zip file or folder so I didn't run your code.

thanks a lot， I have complete your TODOs and got a wonderful vlmaps in real world， I think it should be right！here is the effect！

[image]

[image]

[image]

Tom-Huang, 2023-09-14:
Great job! I am glad that you managed that! I will close the issues

Tom-Huang, 2023-09-14:
fix #13 #14 #12

closed 2023-09-14 by Tom-Huang

soumajm, 2024-04-30:
Hi, we're trying to create VLMaps from a ROS bag. We have synchornized depth, rgb, and pose files. But how to do the transformation is unclear. We used the depth2pc_realworld as advised (with the other TODOs) with the intrinsic matrix but the pc generation from depth images and modified pose as per height. Is there other specifics we'd need to look at?

soumajm, 2024-07-08:
We fixed the issue in the following manner. We recorded the data in the simulator in a manner with the Habitat simulator. Then we could avoid the transformations and use the code from the demo branch as is.