vlmaps/vlmaps #18: poses.txt dataset is missing

hszhoushen, 2023-11-17:
It seems the poses.txt dataset is missing, could you please provide a link to this dataset?

# the structure of the vlmaps_data_dir looks like this
vlmaps_data_dir
  |-5LpN3gDmAk7_1
  |   |-poses.txt
  |-gTV8FGcVJC9_1
  |   |-poses.txt
  |-jh4fc5c5qoQ_1
  |   |-poses.txt
  ...

vardhandongre, 2023-11-24:
Hello @hszhoushen I was working on this and wondering where the pose.txt data is coming from. Were you able to find it out?

wenjhe, 2023-12-07:
link in generate_dataset.py

Tom-Huang, 2023-12-14:
sorry for the delay of answering this. The link should be here: https://drive.google.com/file/d/1KaRi1VnY7C_TT1WckDWxHvP4v3MTNu1a/view?usp=sharing. If you run
```
cd dataset
python generate_dataset.py
```
the code will automatically download the dataset for you. I have just checked the data on google drive and it contains the `poses.txt` file in every scene. If you find the data still incorrect, feel free to reopen this issue.

closed 2023-12-14 by Tom-Huang