vlmaps/vlmaps #19: sudden map creation interrupt

DiTo97, 2023-11-23:
Hi @Tom-Huang,

I have been upgrading our VL mapping pipeline to the refactored codebase, but the map creation process keeps interrupting at a few samples from completion without any apparent error or warning. I even checked the GPU occupancy, but there's plenty of.

The dataset is acquired in real-life with a dual setup of RealSense D435i + T255 using RTAB-map and poses saved in the RGB-D-SLAM format with identifiers. I have written the appropriate creation config files and updated the trajectory loading in the codebase using the [existing function](https://github.com/vlmaps/vlmaps/blob/master/vlmaps/utils/mapping_utils.py#L36) to load real-life poses in that format.

I backtested the map creation with the old codebase and it completed successfully generating a nice 2D BEV VL map, while the new codebase always stops before completion in both 2D and 3D mode.

Any idea why that might happen?

DiTo97, 2023-11-28:
Hi @Tom-Huang. Any updates?

It's becoming critical to solve this for us

Tom-Huang, 2023-12-14:
sorry for the delay. Did you test the code with our dataset? It will be helpful if you can provide a simple script and a few data samples to debug.

DiTo97, 2023-12-14:
No, @Tom-Huang, I did not, but may be worth the try when I come back to vision-language mapping next week.

I can share a small subset of our dataset, with the relevant code and config updates, if you would like to have a look.

closed 2024-07-09 by DiTo97