vlmaps/vlmaps #28: Quaternion

Lazyass66, 2024-01-23:
Can you give some resources to understand and it's effective visualization with habitat api?

Tom-Huang, 2024-01-23:
I am not sure if I understand your question correctly. Do you need some tutorials for habitat-sim? [https://github.com/facebookresearch/habitat-sim/blob/main/examples/tutorials/notebooks/ECCV_2020_Navigation.ipynb](https://github.com/facebookresearch/habitat-sim/blob/main/examples/tutorials/notebooks/ECCV_2020_Navigation.ipynb) this would be a good example and a good tutorial to start with. It is in the official tutorial of habitat-sim.

Lazyass66, 2024-01-23:
I want to understand about the line 71 of clip_mapping_utils , which is r=R.from_quat(quat) , so quat was taken as [3:] from pose file path , my question is what does those 3: values represent,  and what is the reference frame , how can I interactively check that using habitat. Also any resources to understand quaternion and specifically the line 71 , what does R.from_quat(quat) mean ?

Tom-Huang, 2024-01-23:
ah I see. The pose file saves a 7-dim array which is the concatenation of the position (3-dim) and quaternion (4-dim) of the robot pose in habitat-sim. When you see `quat = row[3:]`, we actually take the last 4 values in the array to obtain the quaternion. `R.from_quat(quat)` is a function call from scipy. This is used to convert a quaternion to a 3-by-3 rotation matrix. The reference frame of the pose is habitat-sim's global reference frame. If you want to check the pose in habitat-sim, you can initialize the agent and the simulator using the following code:
```
# sim_cfg should be setup as a dictionary (check cfg in https://github.com/facebookresearch/habitat-sim/blob/main/examples/tutorials/notebooks/ECCV_2020_Navigation.ipynb)
sim = habitat_sim.Simulator(sim_cfg)
agent = sim.initialize_agent(0)
agent_state = habitat_sim.AgentState()
random_pt = sim.pathfinder.get_random_navigable_point()
agent_state.position = random_pt
agent.set_state(agent_state)
obs = sim.get_sensor_observations(0)
# visualize the observations
bgr = cv2.cvtColor(obs["color_sensor"], cv2.COLOR_RGB2BGR)
# resize for less network traffic
if resize:
    bgr = cv2.resize(bgr, (640, 480))
cv2.imshow("rgb", bgr)
cv2.waitKey()
```

Lazyass66, 2024-01-23:
One example of [3:] , which has 4 elements is [0.0, -4.470348358154297e-08, 0.0 , -1 ], so what does each value indicate?

Tom-Huang, 2024-01-23:
the quaternion format is consistent with scipy's definition, basically [x, y, z, w]. This is the input order for `Rotation.from_quat()` function as in [https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.transform.Rotation.from_quat.html](https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.transform.Rotation.from_quat.html).

closed 2024-01-23 by Lazyass66