vlmaps/vlmaps #31: [Question] Real-robot Integration

vcozzolino, 2024-01-30:
Was this tested with a real robot and integrated with the ROS/ROS2 framework? I had a look at the code and it looks like you used only a simulated environment, correct? Or maybe I missed something...

Tom-Huang, 2024-01-30:
It was tested in the real world with the Toyota HSR and with ROS. The idea is to query the map and to generate execution commands (in the form of calling APIs in Python) from a local computer. The interface code like `move_to_pos`, `turn_right`, `move_forward` etc. is implemented such that it returns a list of velocity and execution time pairs. Then we publish these pairs one-by-one to a topic with ROS and the robot subscribes and executes them in sequence. 
So far we haven't released the code for the real robot yet. We will release later when we have time to refactor the code.

vcozzolino, 2024-02-06:
I see, thanks for the answer. So with a real robot, in a real environment, these would be the end-to-end steps to replicate what you do in simulation:

1. Get a bunch of RGBD frames of the real indoor environment.
2. Follow the procedure in the README to create the map and index it.
3. Launch the navigation script and "pilot" the robot.

Is this roughly correct? How would you do step 1 to generate a dataset compatible with VLmaps?

Tom-Huang, 2024-02-16:
The procedure is correct.

vuduc153, 2024-07-02:
@Tom-Huang Hi, I have a question regarding the depth frames captured by a real-world robot
Similar to the process described in the paper, we used a Azure Kinect & RTAB-Map to do an initial scan of the environment and extracted the RGB-D images & optimized poses. I noticed that the depth images captured by the Kinect are rather noisy & have lots of depth "holes". So unlike the depth frames generated from a simulation, sometimes several regions in the images would have no depth information - **and consequently the pixel value would be encoded as 0 in the numpy array**.  
Is this accounted for in the current codebase? Or did you apply some kind of preprocess to the depth images before feeding it to the mapping code during your real-world experience?
Looking forward to hearing back :)
[image]

Tom-Huang, 2024-07-02:
Hi @vuduc153 You can use the code from the main branch. When you back-project the pixels into a 3D point cloud, we call `depth2pc` function, where you can specify the minimum depth value in the argument ([check here](https://github.com/vlmaps/vlmaps/blob/master/vlmaps/utils/mapping_utils.py#L226)). In this way, those "holes" in the depth image indicating 0 depth can be filtered out.

vuduc153, 2024-07-06:
@Tom-Huang Thanks for the prompt answer. We got it to work now!

closed 2024-07-09 by Tom-Huang