vlmaps/vlmaps #33: DISPLAY not detected. For headless systems, compile with --headless for EGL support

kaiiwoo, 2024-02-04:
Hi, thank you for releasing nice work.

However, I encountered the following errors when tried to follow Advanced Usage in README.

1) First error came from running `generate_dataset.py` .  The below is full terminal output.
```bash
(vlmaps) (base) keonwookim@ubuntu:~/vlmaps/dataset$ python generate_dataset.py 
/home/keonwookim/anaconda3/envs/vlmaps/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'generate_dataset.yaml': Defaults list is missing `_self_`. See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
Downloading...
From: https://drive.google.com/uc?id=1KaRi1VnY7C_TT1WckDWxHvP4v3MTNu1a
To: /data/keonwookim/matterport3d/vlmaps_dataset.zip
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 168k/168k [00:00<00:00, 577kB/s]
Archive:  /data/keonwookim/matterport3d/vlmaps_dataset.zip
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/JmbYfDe2QKZ_2/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/JmbYfDe2QKZ_2/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/JmbYfDe2QKZ_2/object_navigation_tasks.json  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/JmbYfDe2QKZ_2/spatial_goal_navigation_tasks.json  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/.DS_Store  
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/YmJkqBEsHnH_1/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/YmJkqBEsHnH_1/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/YmJkqBEsHnH_1/object_navigation_tasks.json  
 extracting: /data/keonwookim/matterport3d/vlmaps_dataset/YmJkqBEsHnH_1/spatial_goal_navigation_tasks.json  
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/jh4fc5c5qoQ_1/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/jh4fc5c5qoQ_1/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/jh4fc5c5qoQ_1/object_navigation_tasks.json  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/jh4fc5c5qoQ_1/spatial_goal_navigation_tasks.json  
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/UwV83HsGsw3_1/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/UwV83HsGsw3_1/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/UwV83HsGsw3_1/object_navigation_tasks.json  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/UwV83HsGsw3_1/spatial_goal_navigation_tasks.json  
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/5LpN3gDmAk7_1/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/5LpN3gDmAk7_1/.DS_Store  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/5LpN3gDmAk7_1/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/5LpN3gDmAk7_1/object_navigation_tasks.json  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/5LpN3gDmAk7_1/spatial_goal_navigation_tasks.json  
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/gTV8FGcVJC9_1/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/gTV8FGcVJC9_1/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/gTV8FGcVJC9_1/object_navigation_tasks.json  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/gTV8FGcVJC9_1/spatial_goal_navigation_tasks.json  
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/Vt2qJdWjCF2_1/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/Vt2qJdWjCF2_1/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/Vt2qJdWjCF2_1/object_navigation_tasks.json  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/Vt2qJdWjCF2_1/spatial_goal_navigation_tasks.json  
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/JmbYfDe2QKZ_1/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/JmbYfDe2QKZ_1/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/JmbYfDe2QKZ_1/object_navigation_tasks.json  
 extracting: /data/keonwookim/matterport3d/vlmaps_dataset/JmbYfDe2QKZ_1/spatial_goal_navigation_tasks.json  
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/mJXqzFtmKg4_1/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/mJXqzFtmKg4_1/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/mJXqzFtmKg4_1/object_navigation_tasks.json  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/mJXqzFtmKg4_1/spatial_goal_navigation_tasks.json  
   creating: /data/keonwookim/matterport3d/vlmaps_dataset/ur6pFq6Qu1A_1/
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/ur6pFq6Qu1A_1/poses.txt  
  inflating: /data/keonwookim/matterport3d/vlmaps_dataset/ur6pFq6Qu1A_1/object_navigation_tasks.json  
 extracting: /data/keonwookim/matterport3d/vlmaps_dataset/ur6pFq6Qu1A_1/spatial_goal_navigation_tasks.json  
Scene 5LpN3gDmAk7_1 :   0%|                                                                                                                               | 0/10 [00:00<?, ?it/s]DISPLAY not detected. For headless systems, compile with --headless for EGL support
```

2) Then I just ignored the error and ran `create_map.py`. This successfully create a single VLMap of scene `5LpN3gDmAk7_1`. But then after running `evaluate_object_goal_navigation.py` , I got same error message like this.
```bash
(vlmaps) keonwookim@ubuntu:~/vlmaps/application/evaluation$ python evaluate_object_goal_navigation.py 
/home/keonwookim/anaconda3/envs/vlmaps/lib/python3.8/site-packages/torch/cuda/__init__.py:138: UserWarning: CUDA initialization: The NVIDIA driver on your system is too old (found version 11060). Please update your GPU driver by downloading and installing a new version from the URL: http://www.nvidia.com/Download/index.aspx Alternatively, go to: https://pytorch.org to install a PyTorch version that has been compiled with your version of the CUDA driver. (Triggered internally at ../c10/cuda/CUDAFunctions.cpp:108.)
  return torch._C._cuda_getDeviceCount() > 0
/data/keonwookim/matterport3d/vlmaps_dataset/5LpN3gDmAk7_1
[21:00:52:066692]:[Metadata] AttributesManagerBase.h(380)::createFromJsonOrDefaultInternal : <Dataset>: Proposing JSON name : default.scene_dataset_config.json from original name : default| This file does not exist.
[21:00:52:066947]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DSolid:capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
[21:00:52:067098]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (capsule3DWireframe:capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
[21:00:52:067145]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneSolid:coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
[21:00:52:067185]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (coneWireframe:coneWireframe_segments_32_halfLen_1.25) created and registered.
[21:00:52:067219]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeSolid:cubeSolid) created and registered.
[21:00:52:067249]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cubeWireframe:cubeWireframe) created and registered.
[21:00:52:067285]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderSolid:cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
[21:00:52:067321]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (cylinderWireframe:cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
[21:00:52:067353]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereSolid:icosphereSolid_subdivs_1) created and registered.
[21:00:52:067385]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (icosphereWireframe:icosphereWireframe_subdivs_1) created and registered.
[21:00:52:067419]:[Metadata] AssetAttributesManager.cpp(121)::createObject : Asset attributes (uvSphereSolid:uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
[... cut ...]
[21:00:52:067948]:[Metadata] MetadataMediator.cpp(175)::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to ./data/default.physics_config.json successfully.
[21:00:52:067968]:[Metadata] MetadataMediator.cpp(66)::setSimulatorConfiguration : Set new simulator config for scene/stage : /data/keonwookim/matterport3d/mp3d/5LpN3gDmAk7/5LpN3gDmAk7.glb and dataset : default which is currently active dataset.
DISPLAY not detected. For headless systems, compile with --headless for EGL support
```

How should I deal with these probs? If these are because of not installing habitat simulator, then what version of the habitat simulator should I install?

Tom-Huang, 2024-02-04:
I think you are running the generation code on a server without a screen. In this case, you should install the habitat-sim with headless option with the following command:
```
conda install habitat-sim headless -c conda-forge -c aihabitat
```
This instruction can be found on the habitat-sim's official web page: [https://github.com/facebookresearch/habitat-sim?tab=readme-ov-file#recommended-conda-packages](https://github.com/facebookresearch/habitat-sim?tab=readme-ov-file#recommended-conda-packages).

kaiiwoo, 2024-02-05:
thanx, I could resolve those issues. But then I ran `create_map.py` again, I met a new error message

```bash
vlmaps) (base) keonwookim@ubuntu:~/vlmaps/application$ python create_map.py 
/home/keonwookim/anaconda3/envs/vlmaps/lib/python3.8/site-packages/torch/cuda/__init__.py:138: UserWarning: CUDA initialization: The NVIDIA driver on your system is too old (found version 11060). Please update your GPU driver by downloading and installing a new version from the URL: http://www.nvidia.com/Download/index.aspx Alternatively, go to: https://pytorch.org to install a PyTorch version that has been compiled with your version of the CUDA driver. (Triggered internally at ../c10/cuda/CUDAFunctions.cpp:108.)
  return torch._C._cuda_getDeviceCount() > 0
Creating map for scene at:  /data/keonwookim/matterport3d/vlmaps_dataset/5LpN3gDmAk7_1
[[ 1.          0.          0.         11.92911911]
 [ 0.          1.          0.         -1.28434992]
 [ 0.          0.          1.         -0.11747979]
 [ 0.          0.          0.          1.        ]]
[[  1.           0.           0.           0.11747979]
 [  0.           1.           0.         -11.92911911]
 [  0.           0.           1.          -1.28434992]
 [  0.           0.           0.           1.        ]]
  0%|                                                   | 0/1159 [01:01<?, ?it/s]
Error executing job with overrides: []
Traceback (most recent call last):
  File "create_map.py", line 20, in main
    vlmap.create_map(data_dirs[config.scene_id])
  File "/home/keonwookim/vlmaps/vlmaps/map/vlmap.py", line 76, in create_map
    self.map_builder.create_mobile_base_map()
  File "/home/keonwookim/vlmaps/vlmaps/map/vlmap_builder.py", line 122, in create_mobile_base_map
    pix_feats = get_lseg_feat(
  File "/home/keonwookim/vlmaps/vlmaps/utils/lseg_utils.py", line 83, in get_lseg_feat
    output, logits = model(pad_crop_img, labels)
  File "/home/keonwookim/anaconda3/envs/vlmaps/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/keonwookim/anaconda3/envs/vlmaps/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/keonwookim/vlmaps/vlmaps/lseg/modules/models/lseg_net.py", line 323, in forward
    logits_per_image = pixel_encoding @ text_features.t()
RuntimeError: expected m1 and m2 to have the same dtype, but got: c10::Half != float

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```

kaiiwoo, 2024-02-07:
I found the error came out from these lines. `pixe_encoding` is `float16`, but `text_features` are `float32`. So I removed `.half()' in line 194 and It worked. 
https://github.com/vlmaps/vlmaps/blob/d394bff70c57823c6dfd20827f7f73696fcd5d78/vlmaps/lseg/modules/models/lseg_net.py#L194-L196

However, a time to build a single map takes about 2 hours... Essentially, I don't understand why this data type mismatch error came out. When I ran `demo.ipynb`, map building was sucessfully exectued without any mismatch error. Are there any other changes in code between master & demo branch?
[image]

Tom-Huang, 2024-02-08:
what is the GPU you used for the map creation? On a GTX TITAN X 12GB, the first scene can be finished within around 20min.

kaiiwoo, 2024-02-08:
It's TITAN Xp (12G), with CUDA version 11.4

Tom-Huang, 2024-02-08:
maybe also check which part is taking the majority time. Is it the get_lseg_feats function or other part of the code? Sometimes the dataloading on the server might be slow due to the network.

Tom-Huang, 2024-02-08:
great! How did you solve this problem? Did you update your CUDA or change the torch version?

kaiiwoo, 2024-02-09:
Yes I changed the workspace to the CUDA 12.2 server. Then I could successfully match the required torch version of vlmap!

Tom-Huang, 2024-02-09:
great, thank you for providing details.

closed 2024-02-09 by Tom-Huang