vlmaps/vlmaps #36: Mapping issue on scene JmbYfDe2QKZ_1 and JmbYfDe2QKZ_2

sairitwik27, 2024-02-13:
I am getting the following issue when I run the object goal navigation code. It occurs only on this scene, for all the other scenes it works well. 

Traceback (most recent call last):
  File "evaluate_object_goal_navigation.py", line 30, in main
    robot.setup_scene(scene_id)
  File "home/vlmaps/vlmaps/robot/habitat_lang_robot.py", line 83, in setup_scene
    self.map.customize_obstacle_map(
  File "/home/vlmaps/vlmaps/map/vlmap.py", line 169, in customize_obstacle_map
    self.obstacles_new_cropped = get_dynamic_obstacles_map_3d(
  File "/home/vlmaps/vlmaps/utils/index_utils.py", line 177, in get_dynamic_obstacles_map_3d
    new_obstacles[obs_pts[:, 0] - rmin, obs_pts[:, 1] - cmin] = 1
IndexError: index 277 is out of bounds for axis 1 with size 277

weiyunpei, 2024-04-09:
hello, have you ever solve the problem? I get the same issue running create_map.py

Tom-Huang, 2024-04-09:
Thank you for your interest in our project. The bug is fixed with this commit now: https://github.com/vlmaps/vlmaps/commit/9dcf2e46a6d7a6b8c3bbefaa4de22d3bab7bd7fd.

closed 2024-04-09 by Tom-Huang