vlmaps/vlmaps #37: Role of shuffle in point cloud reconstruction via depth

kaiiwoo, 2024-02-26:
Hi, it is hard to intuitively understand the roll of mask & shuffle operation in the following snippet. What are these codes work for? 

https://github.com/vlmaps/vlmaps/blob/d394bff70c57823c6dfd20827f7f73696fcd5d78/vlmaps/map/vlmap_builder.py#L273-L279

kaiiwoo, 2024-02-26:
Also, is it OK to set focal length to a half of width? I think is heuristic to compute camera parameter. What's the main reason of computing like this? 
https://github.com/vlmaps/vlmaps/blob/d394bff70c57823c6dfd20827f7f73696fcd5d78/vlmaps/utils/mapping_utils.py#L583-L588

Tom-Huang, 2024-02-26:
> Hi, it is hard to intuitively understand the roll of mask & shuffle operation in the following snippet. What are these codes work for?
> 
> https://github.com/vlmaps/vlmaps/blob/d394bff70c57823c6dfd20827f7f73696fcd5d78/vlmaps/map/vlmap_builder.py#L273-L279

The major reason for shuffling the depth point cloud is to downsample the point cloud randomly. As you can see in [Line 276](https://github.com/vlmaps/vlmaps/blob/d394bff70c57823c6dfd20827f7f73696fcd5d78/vlmaps/map/vlmap_builder.py#L276), we take every `depth_sample_rate` point in the depth point cloud as the subsampled point cloud. If there is no shuffling, the subsampled point cloud might follow a certain pattern. However, the way I implemented it is a bit confusing. Here is a short explanation:
line 273: Get the point cloud backprojected from the depth image (a list of point) and the mask of valid points (a list of Trues and Falses. Points with distance within a certain range are True, else False)
line 274: Get the indices of all points ([0, 1, 2, ..., N])
line 275: Shuffle the indices ([3, 28, 1, ..., ])
line 276: Get the downsampled indices (returned total index number: N / depth_sample_rate)
line 277: Since the point cloud will be downsampled, the valid mask obtained from line 273 is also downsampled.
line 278: Downsample the point cloud.
line 279: Mask the point cloud.

Feel free to ask further questions if you still find it confusing.

Tom-Huang, 2024-02-26:
> Also, is it OK to set focal length to a half of width? I think is heuristic to compute camera parameter. What's the main reason of computing like this?
> 
> https://github.com/vlmaps/vlmaps/blob/d394bff70c57823c6dfd20827f7f73696fcd5d78/vlmaps/utils/mapping_utils.py#L583-L588

You are right! Actually, this is only valid when the horizontal field of view of the image is 90 degrees. That's why we also provide a version of code that computes the intrinsic matrix considering the fov [here](https://github.com/vlmaps/vlmaps/blob/d394bff70c57823c6dfd20827f7f73696fcd5d78/vlmaps/utils/mapping_utils.py#L609). In our collected data in habitat sim, we used a camera with fov=90deg, that's why it is still valid in sim. For other camera setups, you should specify a correct intrinsic matrix. You can specify the camera intrinsic matrix in the config file at [this line](https://github.com/vlmaps/vlmaps/blob/d394bff70c57823c6dfd20827f7f73696fcd5d78/config/map_config/vlmaps.yaml#L10).

kaiiwoo, 2024-02-26:
Thanks for your replies. I resolved Q2 clearly. In Q1, then, what is the physical meaning of the emergence of a pattern in downsampled point clouds? why does it matter?

Tom-Huang, 2024-02-26:
The pattern I mean is something like the following image. For example, if we have a depth image with size 3-by-6. We call [depth2pc](https://github.com/vlmaps/vlmaps/blob/d394bff70c57823c6dfd20827f7f73696fcd5d78/vlmaps/utils/mapping_utils.py#L226) to compute the point cloud. If you take a closer look at how `depth2pc` is implemented, you will find out that the resulting point cloud is ordered as is shown in the image. The number in each pixel refers to the index of the corresponding point in the resulting point cloud. Now if we downsample the point cloud with a depth_sample_rate of 6, 7, 8, we will find that the subsampled point cloud corresponds to the pixels in grey which is not random but follows a certain pattern.
[image]

kaiiwoo, 2024-02-26:
Now I could clearly understand! Thanks for ur kindness!

closed 2024-02-27 by Tom-Huang