vlmaps/vlmaps #38: Issue with  "Vt2qJdWjCF2_1" , while running create_video with this scene.

clarencerat, 2024-03-09:
0%|          | 0/1454 [00:00<?, ?it/s]WARNING:imageio_ffmpeg:IMAGEIO FFMPEG_WRITER WARNING: input image is not divisible by macro_block_size=16, resizing from (1080, 720) to (1088, 720) to ensure video compatibility with most codecs and players. To prevent resizing, make your input image divisible by the macro_block_size or set the macro_block_size to 1 (risking incompatibility).
WARNING:imageio_ffmpeg:IMAGEIO FFMPEG_WRITER WARNING: input image is not divisible by macro_block_size=16, resizing from (1080, 720) to (1088, 720) to ensure video compatibility with most codecs and players. To prevent resizing, make your input image divisible by the macro_block_size or set the macro_block_size to 1 (risking incompatibility).
  0%|          | 1/1454 [00:00<08:09,  2.97it/s]

---------------------------------------------------------------------------

BrokenPipeError                           Traceback (most recent call last)

[/usr/local/lib/python3.10/dist-packages/imageio_ffmpeg/_io.py](https://localhost:8080/#) in write_frames(path, size, pix_fmt_in, pix_fmt_out, fps, quality, bitrate, codec, macro_block_size, ffmpeg_log_level, ffmpeg_timeout, input_params, output_params, audio_path, audio_codec)
    626             try:
--> 627                 p.stdin.write(bb)
    628             except Exception as err:

BrokenPipeError: [Errno 32] Broken pipe

During handling of the above exception, another exception occurred:

OSError                                   Traceback (most recent call last)

4 frames

[<ipython-input-14-76fd8ca1735e>](https://localhost:8080/#) in <cell line: 1>()
----> 1 create_video("/content/Vt2qJdWjCF2_1", "/content/", fps=30)

[<ipython-input-1-f1e2047251af>](https://localhost:8080/#) in create_video(data_dir, output_dir, fps)
     74         depth_color = cv2.applyColorMap(depth_vis, cv2.COLORMAP_JET)
     75 
---> 76         rgb_writer.append_data(rgb)
     77         depth_writer.append_data(depth_color)
     78         pbar.update(1)

[/usr/local/lib/python3.10/dist-packages/imageio/core/format.py](https://localhost:8080/#) in append_data(self, im, meta)
    588             im = asarray(im)
    589             # Call
--> 590             return self._append_data(im, total_meta)
    591 
    592         def set_meta_data(self, meta):

[/usr/local/lib/python3.10/dist-packages/imageio/plugins/ffmpeg.py](https://localhost:8080/#) in _append_data(self, im, meta)
    598 
    599             # Write. Yes, we can send the data in as a numpy array
--> 600             self._write_gen.send(im)
    601 
    602         def set_meta_data(self, meta):

[/usr/local/lib/python3.10/dist-packages/imageio_ffmpeg/_io.py](https://localhost:8080/#) in write_frames(path, size, pix_fmt_in, pix_fmt_out, fps, quality, bitrate, codec, macro_block_size, ffmpeg_log_level, ffmpeg_timeout, input_params, output_params, audio_path, audio_codec)
    632                     "OUTPUT:\n".format(err, cmd_str)
    633                 )
--> 634                 raise IOError(msg)
    635 
    636             nframes += 1

OSError: [Errno 32] Broken pipe

FFMPEG COMMAND:
/usr/bin/ffmpeg -y -f rawvideo -vcodec rawvideo -s 1080x720 -pix_fmt rgb24 -r 30.00 -i - -an -vcodec h264_nvenc -pix_fmt yuv420p -b:v 1000k -vf scale=1088:720 -v info -minrate 500k -maxrate 5000k /content/rgb.mp4

FFMPEG STDERR OUTPUT:

Tom-Huang, 2024-03-09:
are you using the `master` branch to generate the dataset and use the `demo` branch to create the video? It could be because of the rgb image file name mismatch. In the demo dataset, the file names are <scene_name>_00000x.png. In the dataset generated in the master branch, the file names are 00000x.png.
Maybe try to change these two lines in the `create_video` function:
```
rgb_list = sorted(os.listdir(rgb_dir), key=lambda x: int(x.split("_")[-1].split(".")[0]))
depth_list = sorted(os.listdir(depth_dir), key=lambda x: int(x.split("_")[-1].split(".")[0]))
```

to 

```
rgb_list = sorted(os.listdir(rgb_dir))
depth_list = sorted(os.listdir(depth_dir))
```
and try again. I suspect that the image list you pass to ffmpeg is empty.

clarencerat, 2024-03-09:
Hi , I changed those two lines. But still its the same error.