vlmaps/vlmaps #39: In the VLMap Creation section, this error happened

enghish, 2024-03-14:
In the VLMap Creation section, this error happened. Please how can I fix it? Thank you

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
Cell In[24], line 13
     10 import torchvision.transforms as transforms
     11 import clip
---> 13 from utils.clip_mapping_utils import load_pose, load_semantic, load_obj2cls_dict, save_map, cvt_obj_id_2_cls_id, depth2pc, transform_pc, get_sim_cam_mat, pos2grid_id, project_point
     15 from lseg.modules.models.lseg_net import LSegEncNet
     16 from lseg.additional_utils.models import resize_image, pad_image, crop_image

ModuleNotFoundError: No module named 'utils'

Tom-Huang, 2024-03-14:
which branch are you using? It looks like your code is using the `demo` branch. The code in the `master` branch was refactored and many imports in the `demo` branch are no longer valid. If you want to use the `demo.ipynb` Jupyter notebook in the `demo` branch, you should also stay in that branch.

vuduc153, 2024-04-03:
There seems to be a small miss in the Colab notebook that makes it fail to checkout to the `demo` branch which causes this error.
[image]
I think you would have to cd into vlmaps first

Tom-Huang, 2024-04-03:
ah thank you for pointing this out!

enghish, 2024-04-04:
Thank you for your help^^

closed 2024-04-04 by Tom-Huang