vlmaps/vlmaps #40: I don't know where to download pose metadata

ExplosiveElements, 2024-03-16:
I have a question regarding your project that I've been studying on GitHub and your project home page. I haven't been able to locate the pose metadata in either location. Could you kindly advise where I could obtain this data so I can reproduce and learn from your project? Thank you in advance for your assistance.

Tom-Huang, 2024-03-16:
We didn't show the link in the readme but in the generate_dataset.py file. If you take a look at [this line](https://github.com/vlmaps/vlmaps/blob/master/dataset/generate_dataset.py#L71), you can see the link to the goodle drive. If you follow the generate dataset section of the readme by running generate_dataset.py, the data will be automatically downloaded for you given the path you specified in the config.

ExplosiveElements, 2024-03-17:
> We didn't show the link in the readme but in the generate_dataset.py file. If you take a look at [this line](https://github.com/vlmaps/vlmaps/blob/master/dataset/generate_dataset.py#L71), you can [...]

Thank you very much for your detailed explanation, it has been extremely helpful in addressing my issues. I now understand that the linkage to load the data can be found in the generate_dataset.py file. I will follow your advice and am eagerly looking forward to learning more from your project. I appreciate your patience and assistance once again. Best regards!

closed 2024-04-02 by Tom-Huang