vlmaps/vlmaps #43: IndexError when create maps in scence 5LpN3gDmAk7_1

weiyunpei, 2024-04-09:
Hello , thanks a lot for your sharing. But I got some errors when running `applications/create_map.py`

It works well when the `scene_id` gets the value from **1** to **3**. But when I set `scene_id ` to  **0**  in `config/map_creation_cfg.yaml`, I got the issue as follows:
> Creating map for scene at:  /mnt/nas/weiyunpei/vlmaps/data/vlmaps_dataset/5LpN3gDmAk7_1
> [[ 1.          0.          0.         11.92911911]
>  [ 0.          1.          0.         -1.28434992]
>  [ 0.          0.          1.         -0.11747979]
>  [ 0.          0.          0.          1.        ]]
> [[  1.           0.           0.           0.11747979]
>  [  0.           1.           0.         -11.92911911]
>  [  0.           0.           1.          -1.28434992]
>  [  0.           0.           0.           1.        ]]
> [2024-04-09 23:25:30,985][torch.distributed.nn.jit.instantiator][INFO] - Created a temporary directory at /tmp/tmp9iaxtjf3
> [2024-04-09 23:25:30,985][torch.distributed.nn.jit.instantiator][INFO] - Writing /tmp/tmp9iaxtjf3/_remote_module_non_scriptable.py
>   0%|                                                                                       | 0/1159 [00:36<?, ?it/s]
> Error executing job with overrides: []
> Traceback (most recent call last):
>   File "create_map.py", line 22, in main
>     vlmap.create_map(data_dirs[config.scene_id])
>   File "/mnt/nas/weiyunpei/vlmaps/vlmaps/map/vlmap.py", line 76, in create_map
>     self.map_builder.create_mobile_base_map()
>   File "/mnt/nas/weiyunpei/vlmaps/vlmaps/map/vlmap_builder.py", line 167, in create_mobile_base_map
>     grid_feat[max_id] = feat.flatten() * alpha
> IndexError: index 198652 is out of bounds for axis 0 with size 198652

I wonder why it happened just in scene 0, thanks for your time.

Tom-Huang, 2024-04-09:
Thank you for your interest in our project. I think it happened because you already created the vlmap.h5df file before. You can solve it by removing the old vlmap.h5df file and create a new one. Let me know if this works. Thanks!

weiyunpei, 2024-04-10:
> Thank you for your interest in our project. I think it happened because you already created the vlmap.h5df file before. You can solve it by removing the old vlmap.h5df file and create a new one. Let me know if this works. Thanks!

It works well whene I delete the .h5df flie and run `applications/create_map.py` again. Thanks for your help, it really helps a lot.

closed 2024-04-10 by weiyunpei

Raghuveer22, 2024-07-03:
Hey i would like to connect with you regarding config I'm getting error when i'm working with this scene
this is saying maxid is out of bounds when i kept my gs as 1000 and worked when i kept gs as 2000
please i really like to connect with you

Tom-Huang, 2024-07-03:
did you try to delete the `.h5df` file in the output `vlmap` folder and retry? Did you change the `gs` in between creating the map and indexing the map?

Raghuveer22, 2024-07-03:
I will share my findings tomorrow with you sir 
please help me today my gpu is down to run the code again
thanks you for connecting...

Tom-Huang, 2024-07-03:
Sounds good. Let me know when you have the findings.

Raghuveer22, 2024-07-04:
The obstacle map for the scene is generated incorrectly if I run the VLMapBuilder function but the demo folder gives correct obstacle map but its relative to the camera perspective of the first position and not the original 3d coordinates.