vlmaps/vlmaps #46: How to visualize the simulation effect

CYALJH, 2024-04-24:
Hello, I would like to ask you, how to make the simulation of the navigation effect visualization? Because when I ran evaluate object goal navigation.py, I found that after loading the clip model, only an obstacle map was displayed. And then the program stops running. What is the reason for this?I set vis to true in the configuration file

Tom-Huang, 2024-04-24:
Thank you for your interest in our project. We noticed that the problem was due to the fact that we used the old OpenAI API in our code. We just updated the API calls in [llm_utils.py](https://github.com/vlmaps/vlmaps/blob/master/vlmaps/utils/llm_utils.py) and [index_utils.py](https://github.com/vlmaps/vlmaps/blob/master/vlmaps/utils/index_utils.py#L34). The navigation code can visualize properly now.

CYALJH, 2024-04-26:
Hello, thank you for your reply. I had previously modified the model provided by openai and I was using gpt-3.5-turbo-instruct. After I modified the model you provided, I still couldn't realize visualization after running it. Only the obstacle map is shown. As shown in the figure.
[image]
[image]

CYALJH, 2024-04-26:
[image]

Tom-Huang, 2024-04-26:
did you press any key on the keyboard to proceed?

CYALJH, 2024-04-26:
I hit Enter. Nothing happened

Tom-Huang, 2024-04-26:
did you focus on the visualization window "new_obstacles_cropped" and press Enter? Normally, there will be another window popping up showing the contours of obstacles which are used for navigation graph creation.

Tom-Huang, 2024-05-15:
Hi @CYALJH, have you succeeded in visualizing the navigation?

CYALJH, 2024-05-16:
Sorry for the late reply. Yes, I have successfully visualized it. Thank you for your help!

closed 2024-05-16 by Tom-Huang

BIT-TYJ, 2024-06-01:
Hello, I would like to ask if the navigation of VLMaps is implemented in the Habitat simulator, will it show the real-time movement of the robot?