vlmaps/vlmaps #47: Index_map too slow

Bluedask, 2024-05-14:
When I run "python index_map.py", the progarm is taking too much time in the "o3d.visualization.draw_geometries([pcd])" (serveral hours) and it cannot finish function "visualize_rgb_map_3d". Therefore, I cannot input the corresponding query to "input("What is your interested category in this scene?")". Is this really time-consuming or I make some mistakes?
[image]
[image]

Tom-Huang, 2024-05-15:
Hi, thank you for your interest in our project. Have you try to visualize some other point clouds with similar point numbers? Do they also take such a long time?

Bluedask, 2024-06-06:
Sorry to be late, I w

Bluedask, 2024-06-06:
Sorry for the late reply. I have been working on other projects for the past few weeks. I will test your suggestions this week, and if I get any results, I will share them.

Bluedask, 2024-06-18:
Previously, I was using Ubuntu without a desktop environment. After switching to Ubuntu with a desktop environment, everything seems to have normalized. Thank you for your response.

closed 2024-06-18 by Bluedask