vlmaps/vlmaps #48: Question and Urgently in need of help

ChinaXang, 2024-05-15:
I attempted to use an RGBD camera to scan a room and replace the dataset in your project with the obtained data. I wanted to create a local VLMap, but the visualized map effect was poor after running index_map.py.
I create a local VLMap using the following method:
1. Using a machine car equipped with an RGBD camera to scan a circle around the room to obtain a ROSBAG.
2. Extract RGB images, depth information, and poses from ROSBAG
3. Replace RGB images, depth information, and poses in your project
4. Change the camera internal matrix in your project to my camera internal matrix.
5. Run createmap.py.
May I ask if the effect is not good because the data I recorded is not good（Because I have some images that are totally black when extracting depth images from my rosbag）, or do I need to modify some parts of your code。It would be even better if you could provide some suggestions for building a local map。

Tom-Huang, 2024-05-15:
Could you provide an example script or some visualization of running index_map.py?

Tom-Huang, 2024-05-15:
How do you represent your pose? Is it [x, y, z, qx, qy, qz, qw] (the way our code assume) for each row in the poses.txt file?

ChinaXang, 2024-05-15:
[image]
[image]

Tom-Huang, 2024-05-15:
It might be because your quaternion representation is not in our order [qx, qy, qz, qw]. Maybe try to put the 4-th column to the final column.

ChinaXang, 2024-05-15:
OK，I try your idea first，thanks

ChinaXang, 2024-05-15:
[image]

ChinaXang, 2024-05-15:
[image]

ChinaXang, 2024-05-15:
still some problem. maybe the poses is same to yours

ChinaXang, 2024-05-15:
```
import rosbag
import numpy as np

# 读取 rosbag 文件
bag = rosbag.Bag('/home/isqianq/2024-05-13-16-33-10.bag')

# 定义保存位姿信息的文件名
output_file = 'poses.txt'

# 打开文件以写入位姿信息
with open(output_file, 'w') as f:
    # 遍历 rosbag 中的每个消息
    for topic, msg, t in bag.read_messages():
        # 检查是否为位姿信息，你需要根据实际的 rosbag 中的消息类型来修改条件
        if topic == '/zed2i/zed_node/odom':
            # 提取位姿信息，这里假设位姿信息为四元数
            pose = [msg.pose.pose.position.x, msg.pose.pose.position.y, msg.pose.pose.position.z,
                    msg.pose.pose.orientation.x, msg.pose.pose.orientation.y, msg.pose.pose.orientation.z, msg.pose.pose.orientation.w]
            # 将位姿信息写入文件
            f.write('      '.join(map(str, pose)) + '\n')

# 关闭 rosbag 文件
bag.close()
```

Tom-Huang, 2024-05-15:
ok, the file looks good to me. Actually, in our code, we assume that the poses.txt is for the poses of the robot base and you need to specify the transformation from the camera to the base. If your poses are for the camera, you can change [these lines](https://github.com/vlmaps/vlmaps/blob/master/config/map_config/vlmaps.yaml#L6-L9)  to:
```
  base2cam_rot: [1, 0, 0, 0, 1, 0, 0, 0, 1]
  base_forward_axis: [0, 0, 1]
  base_left_axis: [-1, 0, 0]
  base_up_axis: [0, -1, 0]
```
and then try again.

ChinaXang, 2024-05-15:
It is true that my poses is the camera's poses.But it is still bad after i change these lines you said.

ChinaXang, 2024-05-15:
[image]

ChinaXang, 2024-05-15:
Is this due to poor extraction of depth information. Because the depth information in rosbag exists in image format, I need to first extract the depth image and then convert it to npy format. I found that many of the extracted depth images are completely black.The RGB and poses may be fine.
Or is it because the recorded information is not complete enough to scan all the information of the room.

ChinaXang, 2024-05-17:
hello，I think it's probably because the pose information doesn't match yours, because I found that the Z-axis quantity（position z） in my pose data is basically unchanged, but the y-axis quantity（position y） in your pose is unchanged. I think it might be because the orientation of the coordinate system where the pose is located is different. Do you think this has an impact?

vcozzolino, 2024-06-10:
Hi, we are also trying to create a semantic map starting from ROS-collected data. We tried with hundreds of different transformation matrixes and double-checked our data, but the generated map was always totally wrong. I have attached a small sample of our data including RGB, depth, poses, and camera intrinsic. Could you be so kind to have a look at it? Maybe you can find out why it doesn't work.

[image]
[sample.zip](https://github.com/user-attachments/files/15758875/sample.zip)

Tom-Huang, 2024-06-10:
@vcozzolino could you briefly describe your robot setup including whether the camera is z forward, x rightward, y downward, the transform from camera frame to the robot base coordinate frame, and which axis of the robot base coordinate frame is forward, leftward, and downward? That would help understanding how to config the transform.

soumajm, 2024-06-19:
The camera axes is standard with z as forward, x rightward, y downward.

The robot base coordinate frame is ROS style with z upwards, x forward, and y to the left.

We use the following transform -
base2cam_rot = np.array([
    [0, 0, 1,],
    [-1, 0, 0],
    [0, -1, 0]
])

soumajm, 2024-06-26:
Hi thanks for your response. 

I am confused now. The idea is to transform the base pose to the camera pose right for the first frame (and additionally account for the height)? then for next, we have the inverse for registering all future frames to the first one.

Another way of asking would be, if I follow the exact process as in the "demo" branch, just using the transform of the matrix* will be sufficient (and + the camera height along pos[2])? Thanks.

matrix* = 
array([[ 0, -1,  0],
       [ 0,  0, -1],
       [ 1,  0,  0]])

Tom-Huang, 2024-06-26:
> The idea is to transform the base pose to the camera pose right for the first frame (and additionally account for the height)? then for next, we have the inverse for registering all future frames to the first one.

The demo branch treats the initial camera frame as the world coordinate frame and the main branch (refactored code) takes the first robot base coordinate frame as the world frame.

ChinaXang, 2024-06-29:
> Hi thanks for your response.
>
> I am confused now. The idea is to transform the base pose to the camera pose right for the first frame (and additionally account for the height)? then for next, we have [...]

hello，do you solve the problem? My camera axes and robot base coordinate frame are same with you. The first robot base coordinate frame is as the world frame. 
base2cam_rot* =
array([[ 0, -1, 0],
[ 0, 0, -1],
[ 1, 0, 0]])
base_forward_axis: [1, 0, 0]
base_left_axis: [0, 1, 0]
base_up_axis: [0, 0, 1]
But the map is still wrong. 
[image]

ChinaXang, 2024-06-29:
It is the format of poses data. Could you be so kind to have a look at it? 
[image]

ChinaXang, 2024-06-29:
Hello .I see that your team tested in real world with Toyota HSR and ros . When creating the map , do you only change the parameters in the .yaml ?Or we need to modify code.
And there is another map. I found that the approximate shape of the map may be correct, but the details are very poor.
[image]

soumajm, 2024-07-02:
> hello，do you solve the problem? My camera axes and robot base coordinate frame are same with you. The first robot base coordinate frame is as the world frame. base2cam_rot* = array([[ 0, -1, 0], [ [...]

Hi, no it doesn't work.

isqianxy, 2024-07-03:
Hi, Huang. Could you give some advice about the problem? Thanks.

Tom-Huang, 2024-07-03:
I update the code base to support camera-based mapping so that you don't need to setup the relative transformation between base and the camera. I also added a section in README to detail how to use VLMaps with customized dataset: [here](https://github.com/vlmaps/vlmaps?tab=readme-ov-file#application-on-customized-datasets). 

In short, I recommend you to: 
1. save your camera poses in the world coordinate frame in the `poses.txt` file. Each line stores the 16-dimensional pose (flattened 4x4 pose).
2. change the `config/map_cofig/vlmaps.yaml` file: 
  (1) change `pose_info.pose_type` to `"camera_base"`.
  (2) change `pose_info.rot_type` to `"mat"`.
  (3) change `cam_calib_mat` to your camera intrinsics.
In this way, you can ignore all relative transformation between the base frame and the camera frame for creating a 3D VLMap. Just call `python applications/create_map.py` and `python applications/index_map.py` to test it briefly.

If you find the solution useful, please also share with others here and I will close the Issue.

Tom-Huang, 2024-07-03:
If you still can not solve the problem, please reply with a zip file containing the data in the following format:
```
# the structure of the vlmaps_data_dir looks like this
vlmaps_data_dir
  |-customized_scene_1
  |   |-rgb
  |   |  |-000000.png
  |   |  |-000001.png
  |   |-depth
  |   |  |-000000.npy
  |   |  |-000001.npy
  |   |-poses.txt # save the 16-dimensional camera poses in world coordinate frame
  ...
```
and I will test it for you.

isqianxy, 2024-07-04:
ok，thanks for that，I will try it。

ChinaXang, 2024-07-04:
there are some bugs. when i run create_map.py, it is Interrupting in running code. so the map is Incomplete. The error and map are follow
[image]
[image]
[image]

Tom-Huang, 2024-07-04:
@ChinaXang Could you share 3-5 frames of the data for testing?

ChinaXang, 2024-07-04:
hi,huang. Thanks very much.I have solved the problem. we use another dataset called uhumans2 . The map we created looks good. Our poses are base poses .So we used mobile_base to create map. We ignored the translation from base to camera before. So the result looks bad . We are now preparing to record a scene of our laboratory to create A map of the real environment. Hope things go well.

Tom-Huang, 2024-07-04:
It's great that you solve the problem! I will close the issue for now. For further related questions, feel free to reopen it.

Tom-Huang, 2024-07-04:
> there are some bugs. when i run create_map.py, it is Interrupting in running code. so the map is Incomplete. The error and map are follow ![截图 2024-07-04 15-27-58](https://private-user-images.githubusercontent.com/169908354/345737116-2b2e20a7-522f-49d4-8aba-314a00158f09.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAwODY5MzQsIm5iZiI6MTcyMDA4NjYzNCwicGF0aCI6Ii8xNjk5MDgzNTQvMzQ1NzM3MTE2LTJiMmUyMGE3LTUyMmYtNDlkNC04YWJhLTMxNGEwMDE1OGYwOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwNFQwOTUwMzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNWUwZDdlZDk5NzQyZmFjYzBkOTJkNGYwNTVmYjI2ODQ4MGEwYjdhNzA2Y2UwNWM5YzUwNjgwODdkMGE2YjBmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.W6SX3_xovMAutB-8g_Tebsihz5HY3Jsyzyc3-h8YBu0) ![截图 2024-07-04 15-23-22](https://private-user-images.githubusercontent.com/169908354/345737152-65ebd786-9fbd-49da-ad49-3d30923c8d8f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAwODY5MzQsIm5iZiI6MTcyMDA4NjYzNCwicGF0aCI6Ii8xNjk5MDgzNTQvMzQ1NzM3MTUyLTY1ZWJkNzg2LTlmYmQtNDlkYS1hZDQ5LTNkMzA5MjNjOGQ4Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwNFQwOTUwMzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NWM2NThmNmNmY2E2YWU4MTUzZjlhMTliNGI1YmI5ODNhYzVjNTg4ZGM5NDc1NDI1ZjlmOTczZGRlYWQzMDcwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xz6psZKLZdCyzmbZP0bFA51yDL6P-VjJSmMCeKZo0EU) ![截图 2024-07-04 15-23-14](https://private-user-images.githubusercontent.com/169908354/345737226-2a9a273f-1b8b-4532-aa70-33d87a21cd6e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAwODY5MzQsIm5iZiI6MTcyMDA4NjYzNCwicGF0aCI6Ii8xNjk5MDgzNTQvMzQ1NzM3MjI2LTJhOWEyNzNmLTFiOGItNDUzMi1hYTcwLTMzZDg3YTIxY2Q2ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwNFQwOTUwMzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ZTVhMDU4ZDhiMzk5OTNkZDE0NGEwNmQ4MmIwMzc4NzdmZDM0MDUyZTAwZmUwOTM0ZjIxM2I2ZmQ5MmNiYjQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.WuKNeshVlud0VEtzzFB3JLpbMUIKs17XZurFC3hXKwQ)

Did you figure out why this happened? Could you share how you solve it?

closed 2024-07-04 by Tom-Huang

ChinaXang, 2024-07-04:
I do not solve this. I use the code before which do not have camera_base code.

Tom-Huang, 2024-07-09:
> there are some bugs. when i run create_map.py, it is Interrupting in running code. so the map is Incomplete. The error and map are follow ![截图 2024-07-04 [...]

@ChinaXang I found the bug. It turned out that I forgot to remove the line that breaks the for loop when the frame id is larger than 10. It is fixed now.

isqianxy, 2024-07-09:
ok. that is good.