vlmaps/vlmaps #49: object_navigation_tasks.json

xiaolu-luu, 2024-05-24:
I would like to build more test scenarios in other mp3d datasets. May I ask how `object_navigation_tasks.json` and `spatial_goal_navigation_tasks.json` are generated? Thank you very much for your work！

Tom-Huang, 2024-06-03:
It is generated by manual labeling. You can build a map with ground truth semantics and visualize the top-down view of it. Then you can show the top-down map in an interactive viewer and click on it to select the locations while noting down corresponding instructions.

closed 2024-06-03 by Tom-Huang

Heathcliff-saku, 2024-10-17:
Hi, @Tom-Huang 
Could you provide an example of object_navigation_tasks.json? Since the README does not include a description of its format or instructions on generating it, I'm a little unsure. What I know is that it needs to include instruction, objects_info( semantic label, position ), etc.