vlmaps/vlmaps #51: AI2-THOR

xiaolu-luu, 2024-06-02:
I tested in the AI2-THOR  simulation environment, but the effect is very poor, what may be the reason? Can you provide the test code in the AI2-THOR  simulation environment mentioned in the paper? How to collect data in AI2-THOR simulation environment?

[image]

xiaolu-luu, 2024-06-02:
[image]
What might be the cause of this?Why can one wall be aligned, but the side walls can't be aligned？Thank you very much

Tom-Huang, 2024-06-03:
It looks like something is wrong with the transformation and poses. Could you share a code snippet to reproduce this?

xiaolu-luu, 2024-06-05:
`  base2cam_rot: [1, 0, 0, 0, 1, 0, 0, 0, 1]
  base_forward_axis: [0, 0, 1]
  base_left_axis: [-1, 0, 0]
  base_up_axis: [0, -1, 0]`
Originally, I changed the coordinate change to the same as you answered in another question, and later I tried to change it to the following to achieve the alignment effect,But why is this happening?Thanks for your reply!
`  base2cam_rot: [1, 0, 0, 0, 1, 0, 0, 0, 1]
  base_forward_axis: [0, 0, -1]
  base_left_axis: [-1, 0, 0]
  base_up_axis: [0, 1, 0]`
[image]

Tom-Huang, 2024-06-28:
I think it is because that in AI2THOR, the robot base coordinate frame has the same orientation as the camera (z forward, x right, y downward).

Tom-Huang, 2024-06-28:
I tested the following config and it works for AI2THOR with the main branch: ```base2cam_rot: [1, 0, 0, 0, 1, 0, 0, 0, 1] base_forward_axis: [0, 0, -1] base_left_axis: [-1, 0, 0] base_up_axis: [0, 1, 0]```. I will mark this as solved. If there is further question, please reopen the issue.

closed 2024-06-28 by Tom-Huang