vlmaps/vlmaps #53: how to implement in real scene

qingyanghijk, 2024-07-07:
Hello, I replicated your method on the uHumans2 dataset and it worked very well. But when trying to use vlmaps in the real environment, the mapping effect is very poor. I used ZED2i and called its SDK to obtain RGB images, depth npy, and pose. The reason I analyzed may be due to equipment issues, resulting in insufficient quality of depth maps and low accuracy of poses,
Do you use sensors to obtain real-world data and create maps? Can you give me some suggestions to get a better point cloud map?

Tom-Huang, 2024-07-07:
It's great that you can replicate the method in another dataset. VLMaps is sensitive to depth and odometry noise and that is the reason why the mapping results in the real world are usually inferior. However, there are several ways you can improve the results:
1. Limit the depth range of pixels that are back-projected. Depth sensors usually generate less accurate depth values at long distance. So you can change the [max_depth argument of self._backproject_depth](https://github.com/vlmaps/vlmaps/blob/master/vlmaps/map/vlmap_builder.py#L132) to a lower value like 4.
2. Use other methods to get the odometry. In the paper, we mentioned that we used RTABMap to get the odometry from the RGB-D sequence. However, the results are suboptimal. Given the advancement in SLAM and visual odometry methods nowadays, you can use other methods like ORBSLAM3 etc. to get more accurate odometry. Another method is to use other sensor on the robot to get the odometry. For example, most of the mobile robot has a 2D LiDAR which can give you accurate 3DOF pose on the navigation plane. With the extrinsics between the 2D LiDAR and the RGB-D camera, you can derive the 6DOF odometry of the camera given the 3DOF LiDAR poses. This is also mentioned in our follow-up paper [AVLMaps](https://avlmaps.github.io/).

I hope these details can help you.

ChinaXang, 2024-07-09:
Hi，Huang.We create a real world map. We find that the map removed information with a height greater than one value. You can see the photos follwed.
[image]
[image]
You can see the window in the map has been cut off. Can you give some advice?

Tom-Huang, 2024-07-09:
@ChinaXang By default VLMaps is used for 2D planning and the created 3D map has the default height of the camera height ([see here](https://github.com/vlmaps/vlmaps/blob/master/vlmaps/map/vlmap_builder.py#L94)). However, you can change this line's `camera_height` argument to a larger number to support a 3D map with larger height value.

isqianxy, 2024-07-09:
The map is under camera_height=1.5m. I try to change the camera_height to 1.8,2,even 3m. But these maps are nearly same. So i wonder if  the map removed information with a height greater than one value.

Tom-Huang, 2024-07-09:
I think you should keep `camera_height` in the config file to be your actual camera height (by default 1.5). You should set a different height value for the input to `self._init_map` method's first argument which is defined as `map_height`. For now, you should hard code a value at [this line](https://github.com/vlmaps/vlmaps/blob/master/vlmaps/map/vlmap_builder.py#L94).

Jessy-Huang, 2024-07-10:
> Hi，Huang.We create a real world map. We find that the map removed information with a height greater than one value. You can see the photos follwed. ![截图 2024-07-09 20-07-47](https://private-user-images.githubusercontent.com/169908354/346956619-a2745514-cc3a-4e13-a45a-0d50fb1031ad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1ODA1NjcsIm5iZiI6MTcyMDU4MDI2NywicGF0aCI6Ii8xNjk5MDgzNTQvMzQ2OTU2NjE5LWEyNzQ1NTE0LWNjM2EtNGUxMy1hNDVhLTBkNTBmYjEwMzFhZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxMFQwMjU3NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zY2ZkODFkYmQwYWIzYmNlYTExMzgwNzlkNTEyZWExMjE1ZDI1NjQzY2FmMGY3YjE1OWQ2NGM4MzA0ZTVmM2UxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XreePI9Y5bUgy55QuVca4xyAz8G40bLsPFJV5tibilo) ![000310](https://private-user-images.githubusercontent.com/169908354/346956744-8c8a0304-0795-4e75-8d40-c31adba45695.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1ODA1NjcsIm5iZiI6MTcyMDU4MDI2NywicGF0aCI6Ii8xNjk5MDgzNTQvMzQ2OTU2NzQ0LThjOGEwMzA0LTA3OTUtNGU3NS04ZDQwLWMzMWFkYmE0NTY5NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxMFQwMjU3NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MmZmZGI4M2ZmMzM3MGJjODVmMmMwZjE4YjA2NTRkMjRhMjkwNWE2ZWZmZTE3OTU0MjBjY2UxYjhkY2IwODUyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Sfwf0dALtLXzkmyChMxJiLHugWAgRqH3DsBHFFC_bbA) You can see the window in the map has been cut off. Can you give some advice?

Could you please provide more details on how you achieved good reconstruction results in a real-world environment?

qingyanghijk, 2024-07-10:
We used the ZED 2i stereo camera as the raw input for the sensor. Firstly, we attempted to directly call Zed's API to obtain depth npy, RGB images, and pose, but this resulted in very inaccurate pose estimation (directly calling the API only used a visual odometer, which is not robust). Then we tried to subscribe to the zed node topic under ROS, possibly because the camera under ROS fused multiple sensors to obtain the pose, which achieved good results.
there are several ways you can improve the results:：
1. Try to improve the accuracy of depth images and poses as much as possible. If your sensor may also adjust the output mode, set it to the best (we set the depth_mode in zed to "NERUAL").
2. You can try using the ROS mode of the sensor to save data, which may be optimized.
3. Ensure that your various coordinate transformation matrices are correct.
4. Do not conduct experiments in a cluttered room. Our experimental results in another room were also relatively bad. :)

ChinaXang, 2024-07-10:
> I think you should keep `camera_height` in the config file to be your actual camera height (by default 1.5). You should set a different height value for the input to `self._init_map` method's first [...]

yeah.Your  has taken effect. Thanks for that.

Jominsu99, 2025-02-17:
> Hello, I replicated your method on the uHumans2 dataset and it worked very well. But when trying to use vlmaps in the real environment, the mapping effect is very poor. I used ZED2i and called its [...]

Hi, how can I know how to run VLMaps with uHumans2 dataset?