vlmaps/vlmaps #58: Does system overview figure (Fig. 3) from paper contains incorrect information?

BashMocha, 2024-09-22:
I'm trying to understand what's going on underneath the hood by reading the methodology section in your paper.

As far as I understood, the LSeg visual encoder maps an image (the embedding of each pixel lies in the CLIP feature space), and the embeddings from the visual encoder are stored in a matrix with the corresponding locations in the grid map. If I understand correctly, this is what happens on the left side of Fig. 3 (VLMap Creation).

When I look at the <i>Localizing Open-Vocabulary Landmarks</i> Section B., it says,

<i>... we apply the pre-trained CLIP text encoder [10] to convert such a list of texts into a list of vector embeddings  [e0,e1,...,eM ], e∈R ...</i>

I found the corresponding function/lines in the repository:

https://github.com/vlmaps/vlmaps/blob/58060f97239074338ab419a2090d43fa752d724d/vlmaps/utils/clip_utils.py#L133

https://github.com/vlmaps/vlmaps/blob/bc79b26a577e5a9408f86e45e5c16530ca80f867/examples/clip_mapping_lseg_from_scratch_batch.py#L50

However, on the right side of Fig. 3 (Landmark Indexing), I see that the Open-Vocabulary Label Set is fed into the LSeg Text Encoder instead of CLIP, and the embedding representations in the figure match with those in Section B:

[image]

<i>... we apply the pre-trained CLIP text encoder [10] to convert such a list of texts into a list of vector embeddings  <ins>[e0,e1,...,eM ], e∈R</ins> ...</i>

I'm a little confused here. Shouldn't the Landmark Indexing section of Figure 3 have the CLIP Text Encoder instead of the LSeg Text Encoder?

Would be much appreciated if you could enlighten me on this issue.
Thank you for your time.

Tom-Huang, 2024-10-04:
Hi @BashMocha , sorry for the delay of reply. 
> However, on the right side of Fig. 3 (Landmark Indexing), I see that the Open-Vocabulary Label Set is fed into the LSeg Text Encoder instead of CLIP, and the embedding representations in the figure match with those in Section B:
Actually, LSeg freezes the CLIP text encoder during training. Therefore, the text encoders from LSeg and from CLIP are the same. So you can either use the text encoder from LSeg or use that from CLIP. In our implementation, we choose to use the one from CLIP. Sorry for the confusion here.

BashMocha, 2024-10-04:
Thank you for the explanation. I'm closing the issue now.

closed 2024-10-04 by BashMocha