vlmaps/vlmaps #59: Link to download the dataset

Athirakr94, 2024-10-04:
Dear author I have followed your repository instructions to see the jupyter set up and now started with active development side. I have downloaded , extracted  and configured the mp3d dataset but then there is vlmaps dataset downloaded we need to configure. Kindly point me from where to download the same. I have tried with the package downloaded while executing jupyter in vmlaps/datasets, but there is no poses.txt in its subfolders. Pls help

Tom-Huang, 2024-10-04:
Hi thank you for your interest in our project. When you run `python generate_dataset.py`, it will automatically download the vlmaps dataset and unzip it for you under the directory you configure under `vlmaps_data_dir` in `config/data_paths/default.yaml`. If you look into the `generate_dataset.py` file, you can see that [this line](https://github.com/vlmaps/vlmaps/blob/master/dataset/generate_dataset.py#L71-L76) download the vlmaps dataset for you. If you want to download it yourself manually, you can paste the URL ([https://drive.google.com/file/d/1KaRi1VnY7C_TT1WckDWxHvP4v3MTNu1a/view?usp=sharing](https://drive.google.com/file/d/1KaRi1VnY7C_TT1WckDWxHvP4v3MTNu1a/view?usp=sharing)) to your browser to download.

Athirakr94, 2024-10-05:
Thanks for that

closed 2024-10-05 by Athirakr94