vlmaps/vlmaps #60: Unable to test navigation

Athirakr94, 2024-10-14:
Hi authors, I have tried closely following the readme, I was able to create and index the map. Now while trying to test navigation, in the vlmaps_dataset folder I have folders for each scene ID but nothing related to results dir there. I am getting no such path issue. Kindly help me resolve. 
FileNotFoundError: [Errno 2] No such file or directory: '/home/athira/vlmaps_dataset/5LpN3gDmAk7_1/vlmap_obj_nav_results'
I tried adding a line os.mkdir() but as the contents is None I am getting a divide by zero error .

Athirakr94, 2024-10-15:
Hi authors Thanks for the support offered in my previous issues,
In the snippet of evaluation/compute_object_navigation... file we can see os.listdir(metric_save_dir), but for me the folder is empty. so n_tot_tasks=0 and it raise an issue. Kindly help
[image]

Tom-Huang, 2024-10-15:
Hi @Athirakr94 thank you for your interest in our project. You need to run [evaluate_object_goal_navigation.py](https://github.com/vlmaps/vlmaps/blob/master/application/evaluation/evaluate_object_goal_navigation.py) or [evaluate_spatial_goal_navigation.py](https://github.com/vlmaps/vlmaps/blob/master/application/evaluation/evaluate_spatial_goal_navigation.py) before calling [compute_object_goal_navigation_metrics.py](https://github.com/vlmaps/vlmaps/blob/master/application/evaluation/compute_object_goal_navigation_metrics.py) or [compute_spatial_goal_navigation_metrics.py](https://github.com/vlmaps/vlmaps/blob/master/application/evaluation/compute_spatial_goal_navigation_metrics.py).

Athirakr94, 2024-10-17:
Hi @Tom-Huang, Thanks for your reply. I have tried that, in both cases while running I can see a window with new obstacle id cropped. Later nothing happens in terminal or no more window popped up. I tried keeping the code to run over night, still its not getting completed. May I know how to ensure the code is working. Anything more to be done. Pls help.

Athirakr94, 2024-10-21:
Hi Author,
May I know the compute you had to test this navigation. Actually I trying from my laptop with RTX 3060. I doubt whether this is a compute issue. Pls respond.

Tom-Huang, 2024-10-21:
The hardware should not be a problem.

>  I can see a window with new obstacle id cropped

what do you mean by that? Is it a window showing the obstacle map where black areas indicate obstacles and white areas are free spaces? Did you try to press some keys when you saw the obstacle map show up? You might have turned on the visualization. You can change [this line](https://github.com/vlmaps/vlmaps/blob/master/config/object_goal_navigation_cfg.yaml#L8) in the config file to `False` and try again.

Athirakr94, 2024-10-25:
Thanks for that. I am able to run the script without error.  But the subgoal success rates are coming as zero for object navigation and for spatial navigation getting an error that self.subgoal_success_rate is  not an attribute. It seems to be not defined in the script. I am trying to obtain the metrics reported in the paper. But need help here.