vlmaps/vlmaps #68: How to obtain object_navigation_tasks.json?

resecube, 2025-03-01:
Hi,

I am currently working with your project and noticed the file object_navigation_tasks.json. I would like to ask:

How was this task description file obtained?
Is this an open-source dataset, or was it specifically created by your team?
If it is publicly available, could you provide the source or method to generate it?
Thanks in advance for your help!

BashMocha, 2025-03-01:
You can obtain object and spatial navigation task files (per scene) by running the `generate_dataset.py` script inside the `dataset` directory.

It downloads the instruction files from this link:
https://drive.google.com/file/d/1KaRi1VnY7C_TT1WckDWxHvP4v3MTNu1a/view?usp=sharing

Tom-Huang, 2025-03-01:
@resecube Thanks for your interest in our project! The `object_navigation_tasks.json` was created by our team by manually labeling the goal positions corresponding to the instructions. We created that by showing the bird-eye-view (BEV) of the scene with matplotlib and interactively selecting positions on the BEV as starting positions and GT goal positions. Meanwhile, we wrote down the instructions matching the selected positions based on the surrounding objects in the BEV. 

As @BashMocha mentioned (thank you very much), you can run the `generate_dataset.py` to download the data automatically.

Tom-Huang, 2025-03-02:
please close the issue if you think your questions are answered. Thanks a lot!